Lora-net/lr1110_driver #2: [question] hal implementation

chegewara, 2020-08-25:
Hi,
i am doing hal implementation for mcu and i see something worth to ask.
For example here is get get version code that is passing 4 bytes buffer to get version:
https://github.com/Lora-net/lr1110_driver/blob/master/src/lr1110_system.c#L181-L201

but from specification (page 21) we can see that response is 5 bytes length, because of stat1:
https://www.mouser.ca/pdfDocs/UM_LR1110_W_APP_V10.pdf

My question is:
is it on purpose and we have to allocate another buffer with 1 extra byte (maybe 2 if stat1 and stat2 are in response) or maybe it would be better to have that extra byte/s already included in LR1110 main library to simplify HAL code and to avoid creating extra buffer for it?

Thanks

vytvir, 2020-10-12:
> is it on purpose and we have to allocate another buffer with 1 extra byte

I believe this is done by design. Looking at the example `lr1110_hal.c` implementation [here](https://github.com/Lora-net/lr1110_evk/blob/09b3cf93ca1c49adce3c967058cd8012161b39d2/embedded/application/src/lr1110_hal.c#L62) - a dummy byte is clocked out (basically a NOP), therefore the Stat1 register is ignored.

It seems to me that they wanted to separate the logic of checking the _status_ of command execution and the results of the execution themselves.

> (maybe 2 if stat1 and stat2 are in response)

Unless I am mistaken, they're clocked out with the command. (Full-duplex SPI communication)

chegewara, 2020-10-13:
In that case what is the point of implementing stat1 and stat2 in LR1110 firmware just to:
1) ignore it in response
2) add 1 extra spi transaction with dummy byte
If the purpose is to separate status from command it can be done by not sending stats and it will remove overhead on both, LR1110 and MCU.

Actually i dont have to create additional buffer in hal read/write, what i have to do is to send all requests like read RX data with buffer 1/2 bytes bigger than RX-ed data and request 1 or 2 bytes more, depending on function.

closed 2020-10-13 by chegewara

vytvir, 2020-10-13:
The `Stat1` and `Stat2` bytes are clocked out _while_ you are transmitting a command to the device. You would need a full-duplex SPI to make use of this, not all devices support this, which is likely why it is implemented as it is. Not to mention that you _can_ ignore the `Stat1` and `Stat2` bytes completely, they are _optional_.

> Actually i dont have to create additional buffer in hal read/write, what i have to do is to send all requests like read RX data with buffer 1/2 bytes bigger than RX-ed data and request 1 or 2 bytes more, depending on function.

Or, you can clock out a single byte to ignore the `Stat1` field or, optionally, receive 1 byte of data and check against the `Stat1` for result validity as it will tell you if the command was successful or not. This, again, is quite optional as it is possible to validate the data using other methods.

Mind you, I am not a developer of the `lr1110_drivers` and I do think they could have been done better. That said, they're still very concisely written and it's a breath of fresh air compared to some of the other drivers I get to deal with day to day. A better medium for discussing the drivers themselves would be the [LoRa forums - LR1110 subsection](https://lora-developers.semtech.com/knowledge-base/forum/viewforum/39/). Alternatively you can modify the drivers and:

1. Create a pull-request for the developers to be aware of other methods of doing this or
2. Post your 'patch' in the forums mentioned above

matthieuantoine, 2020-10-13:
@vytvir - thanks for you active contribution on this thread!

> I do think they could have been done better

We are more than open to feedback - do not hesitate to create an issue if there is anything you think that can be improved.

@chegewara - we'll work with the team on a way to improve this implementation and let you know when it is ready.