Lora-net/lr1110_driver #3: Failing to update GNSS Almanac, Almanac too old

hkdot, 2020-10-21:
Hi, 

I am currently working on implementing the GNSS part of LR1110. We are not able to do assisted scan mode because the almanac is too old, so we decided to update it. 

We got the Almanac from LoRacloud and proceed to write to LR1110 device through; `lr1110_gnss_almanac_full_update(&LR1110, buffer)`
After that, I try to calculate the CRC and get different value than expected. 
I then try to do assisted scan and get nav message 0 and response almanac too old. 
This is how I setup assisted scan mode
` const lr1110_gnss_date_t gnss_time = rtc_get_timestamp();
    uint8_t inter_cap_delay_s = 30;
    uint8_t NbSvMax = 10;
    uint32_t crc = 0x496D51D0;
    lr1110_gnss_solver_assistance_position_t gnss_position = { 0 };
    lr1110_gnss_constellation_mask_t constellation = LR1110_GNSS_GPS_MASK | LR1110_GNSS_BEIDOU_MASK;
    checkAlmanacUpdate(crc);
    if( !CheckAndStoreAlmanacAge( GNSS_LIMIT_ALMANAC_AGE_DAYS ) )
    {
        NRF_LOG_INFO( "Almanac is too old ! (> %u days)\n", GNSS_LIMIT_ALMANAC_AGE_DAYS );
        
    }
    lr1110_gnss_set_assistance_position(&LR1110, &gnss_position);
    lr1110_gnss_set_constellations_to_use(&LR1110, constellation);
    lr1110_gnss_set_scan_mode(&LR1110, LR1110_GNSS_DOUBLE_SCAN_MODE, &inter_cap_delay_s);
    lr1110_gnss_scan_assisted(&LR1110, gnss_time, LR1110_GNSS_OPTION_BEST_EFFORT,
        LR1110_GNSS_BIT_CHANGE_MASK | LR1110_GNSS_DOPPLER_MASK | LR1110_GNSS_IRQ_PSEUDO_RANGE_MASK, NbSvMax);`

Have I missed something? 
Would appreciate any help or guidance on how to properly update the almanac and do the assisted scanning (if it is wrong).

Best Regards,
Hamza

mverdy, 2020-10-21:
Hi Hamza !
I suspect an error occurs during the Almanac Update process. Let's first check the input buffer is correct.

Can you confirm `buffer` is a 2580-bytes long array? You may have obtained it with operation equivalent to:

```bash
$ curl -s -X GET -H "Authorization: $DAS_TOKEN" -H "Accept: application/json"  "https://das.loracloud.com/api/v1/almanac/full" | jq -r .result.almanac_image | base64 -d > almanac.bin
```
Where `$DAS_TOKEN` contains you DAS authentication token. `almanac.bin` would then contain the almanac under binary format. This is the 2580-byte long buffer to be transmitted to the `lr1110_gnss_almanac_full_update` call.

hkdot, 2020-10-22:
Hi, 

We are actually getting the almanac in similar way, instead of api/v1/almanac we are using api/v3/almanac because v1 doesn't respond. I have attached the almanac we receive, is it correct? 

Best Regards,
Hamza
[almanac.bin.zip](https://github.com/Lora-net/lr1110_driver/files/5419310/almanac.bin.zip)

hkdot, 2020-10-22:
Hi again, 

We updated to transciever firmware version to 0x0303 according to lr1110_updater_tool. 
We updated the almanac through the lr1110_evk_demo_app through the command prompt with almanac update command. The almanac update was successful but when we check again through our code, we get that the almanac is too old and the crc is not changed. 

I'm bit confused.

mverdy, 2020-10-22:
Hi,

I checked the almanac binary you shared and it looks good to me.
If you run the Demo GNSS Assisted from the lr1110_evk_demo_app, does it display the "Almanac too old" message also?

If the demo from the lr1110_evk_demo_app is working (without displaying the message "Almanac too old") then a possible issue comes from the check of the almanac.

hkdot, 2020-10-23:
Hi, 

We just tested with the lr1110_evk_demo_app and it seems to be working (thus not receiving good values). So we have pinpointed the problem to our code on our MCU. 

We are using LR1110 Drivers version 2 (the release before this current one). We are executing the following code for GNSS Assistance:
`const lr1110_gnss_date_t gnss_time = rtc_get_timestamp();
    uint8_t inter_cap_delay_s = 50;
    uint8_t NbSvMax = 255;
    uint32_t crc = 0xA00EB2C9;
    lr1110_gnss_solver_assistance_position_t gnss_position = { 0 };
    GnssHelperAlmanacDetails_t almanac_ages_crc = { 0 };
    gnss_position.latitude = 37.547907;
    gnss_position.longitude = 127.044128;

    lr1110_gnss_constellation_mask_t constellation = LR1110_GNSS_GPS_MASK | LR1110_GNSS_BEIDOU_MASK;
    if(checkAlmanacUpdate(crc) == true)
    {
        NRF_LOG_INFO("CRC Matching");
    }
   
    //if( !CheckAndStoreAlmanacAge( GNSS_LIMIT_ALMANAC_AGE_DAYS ) )
    //{
    //    NRF_LOG_INFO( "Almanac is too old ! (> %u days)\n", GNSS_LIMIT_ALMANAC_AGE_DAYS );
        
    //}
    lr1110_gnss_set_assistance_position(&LR1110, &gnss_position);
    lr1110_gnss_set_constellations_to_use(&LR1110, constellation);
    lr1110_gnss_set_scan_mode(&LR1110, LR1110_GNSS_DOUBLE_SCAN_MODE, &inter_cap_delay_s);
    lr1110_gnss_scan_assisted(&LR1110, gnss_time, LR1110_GNSS_OPTION_BEST_EFFORT, LR1110_GNSS_BIT_CHANGE_MASK | LR1110_GNSS_DOPPLER_MASK | LR1110_GNSS_IRQ_PSEUDO_RANGE_MASK, NbSvMax);

on Scan done 

uint8_t                          nb_sv_detected;    
    lr1110_gnss_detected_satellite_t    list_of_sv[30] = {0};
    uint16_t result_size;
    uint8_t* result_buffer;
    lr1110_gnss_get_result_size(&LR1110, &result_size);
    NRF_LOG_INFO("Result Size %d", result_size);

    lr1110_gnss_read_results(&LR1110, result_buffer, result_size);
    NRF_LOG_HEXDUMP_INFO(result_buffer, result_size);
    if(result.nav_message.size <= GNSS_LR1110_NAV_MESSAGE_MAX_LENGTH)
    {
        lr1110_gnss_read_results(&LR1110, result.nav_message.message, result.nav_message.size);
    }
    lr1110_gnss_get_nb_detected_satellites(&LR1110, &nb_sv_detected);
    NRF_LOG_INFO("Sv detected %d \n", nb_sv_detected);
    lr1110_gnss_get_detected_satellites(&LR1110, nb_sv_detected, list_of_sv);
    lr1110_gnss_get_timings(&LR1110, &result.timings);
    
    NRF_LOG_INFO("Nav Message %d \n", result.nav_message.message[0]);
    if(result.nav_message.message[0] ==LR1110_GNSS_DESTINATION_SOLVER)
    {
        lr1110_gnss_get_nb_detected_satellites(&LR1110, &nb_sv_detected);
        NRF_LOG_INFO("Sv detected %d \n", nb_sv_detected);
        lr1110_gnss_get_detected_satellites(&LR1110, nb_sv_detected, list_of_sv);
        lr1110_gnss_get_timings(&LR1110, &result.timings);
    }
    else if (result.nav_message.message[0] ==LR1110_GNSS_DESTINATION_HOST)
    {
        GetHostErrorFromResult(&result.nav_message);
    }`

This code is based on user manual and the lr1110 evk source code. We skipped the part to check for Almanac age since the CRC matched and it kept giving error on Almanac age. 
Could you please assist us on whether the code we are writing is correct? 

Best Regards,
Hamza

mverdy, 2020-10-28:
Hi Hamza,

You mentioned at the beginning of your message that the EVK demo app is not returning correct value. Could you extend that point? Particularly, what element are you referring to? And what is the value returned, and the value you were expected?

I think you are not using the Double scan correctly. Actually, when using the GNSS double scan, it is up to the host micro-controller to start the second scan at the correct instant by calling the `lr1110_gnss_scan_continuous` API.
In your case, I suggest you to use the Single Scan. To do, replace the `lr1110_gnss_set_scan_mode` with the following:

```c
lr1110_gnss_set_scan_mode(&LR1110, LR1110_GNSS_SINGLE_SCAN_MODE, &inter_cap_delay_s);
```

Note that the `inter_cap_delay_s` is an output argument. The API gives you the number of second to wait between the call of `lr1110_gnss_scan_assisted` and the call of `lr1110_gnss_scan_continuous` when doing *Double Scan*. In the case of *Single Scan* the value should be 0.

Note also that the value provided to variable `NbSvMax` is incorrect. The maximal value is 128. You can use value 0 to indicate the LR1110 to return all satellite detected.

Could you also copy/paste the result buffer returned by the call `lr1110_gnss_read_results`?

The TCXO is configured with the call `lr1110_system_set_tcxo_mode`.

For the LF clock configuration, does it works with `lr1110_system_cfg_lfclk( &LR1110 , LR1110_SYSTEM_LFCLK_XTAL, true );`?

hkdot, 2020-10-29:
Hi, 

The values on EVK Demo app were related to satellites found and then were not able to accurately solve position, I think it is because we were indoors and our external GNSS (which is placed near the window) were not detecting enough satellites, not clear enough sky view. We tested it again and got better result! 

Ohh okay! Thanks for the clarification! 

I just tried with single scan and still having the same issue. I think it is because of the TCXO configuration. For the LF clock configuration we do `lr1110_system_cfg_lfclk(&LR1110, LR1110_SYSTEM_LFCLK_XTAL, true); `. The LF clock config messes with the application and when Radio try to set Lora Modem Packet, it gets stuck in lr1110_hal_write where busy pin is triggered and does not go low... 

The result buffer from 'lr1110_gnss_read_results' shows 0s and result size is 2. Number of detected satellites is 0. I did the testing in parallel with the EVK demo app which showed between 5-12 satellites and gave the location.

Is my configuration overall correct?

Regards,
Hamza

mverdy, 2020-11-03:
Hi,

The result you get with the lr1110_gnss_read_results is unexpected (as you mentioned). Are you waiting for the GNSS scan done interrupt before calling lr1110_gnss_read_results? When you detect the raise of the IRQ line, you must fetch the IRQ register by calling lr1110_system_get_status, and check that the bit for GNSS_SCAN_DONE is raised. After that you must clear the IRQ by calling lr1110_system_clear_irq_status and then call lr1110_gnss_read_results. It should return a buffer that starts with `0101`.

hkdot, 2020-11-03:
Hi, 

I'm doing exactly that with the IRQ and fetching data. It is working perfectly fine with WiFi scan and LoRa. The GNSS SCAN Done is pointing to a function which is executed when the GNSS scan is finished, so I am reading the result but I don't think the GNSS Scan is ever executing properly. I think the problem lays in lr1110_system_cfg_lfclk(&LR1110, LR1110_SYSTEM_LFCLK_XTAL, true), is there another way to try GNSS without calling that API? Because I was able to identify that some functionality stopped working and the busy pin was always high which didn't allow the application to proceed.

matthieuantoine, 2021-01-13:
Hi,

`lr1110_system_cfg_lfclk` is just here to enable the external 32kHz crystal to get a better accuracy. Did you try to comment this line and launch the test again?

Regards,
_Matthieu

jamesl-dm, 2021-01-22:
For what its worth, my boards fail to wake up from warm sleep mode after lr1110_system_cfg_lfclk(&LR1110, LR1110_SYSTEM_LFCLK_XTAL, true), unless the warm sleep includes a timed wakeup.

matthieuantoine, 2021-04-07:
@jamesl-dm - that's a limitation of the chip.

If you put your chip in sleep mode without a timed wakeup, you have to make sure that the LFClock is configured as internal RC (i.e. calling lr1110_system_cfg_lfclk(&LR1110, LR1110_SYSTEM_LFCLK_XTAL, false).

closed 2021-08-22 by matthieuantoine