Lora-net/lr1110_driver #4: lr1110_wifi_scan_time_limit sometimes doubles the expected scan time

jamesl-dm, 2020-10-23:
Hi,

I'm trying the new time limited wifi scans on a chip returning version 0x21010303.
I've noticed that when:

((signal_type == LR1110_WIFI_TYPE_SCAN_B_G_N) && (scan_mode != LR1110_WIFI_SCAN_MODE_FULL_BEACON))

the measured timeout is almost always double the supplied value.
In all other circumstances, it works as expected.

jamesl-dm, 2020-10-23:
Both the _timeout_per_scan_ms_ and _timeout_per_channel_ms_ seem to be affected.

mverdy, 2020-11-18:
Hi @jamesl-dm,

The duration is expected to double as setting the `signal_type` to `LR1110_WIFI_TYPE_SCAN_B_G_N` makes the chip to scan twice per channels (once for Wi-Fi type B and once for Wi-Fi type G/N).

Actually, the maximum expected duration Wi-Fi scan with time limit API is `timeout_per_channel_ms * n_channels * n_wifi_types`.
`n_wifi_types` is `1` for `LR1110_WIFI_TYPE_SCAN_B`, `LR1110_WIFI_TYPE_SCAN_G` and `LR1110_WIFI_TYPE_SCAN_N`,
`n_wifi_types` is `2` for `LR1110_WIFI_TYPE_SCAN_B_G_N`.

I close this ticket, but don't hesitate to re-open it if you are not satisfied with the answer, or if you still suspect an unexpected behavior.

closed 2020-11-18 by mverdy

jamesl-dm, 2020-11-18:
Thanks - makes sense. You should probably update the function documentation to reflect that. Especially since setting the scan mode to BGN doesn't always double the scan time (if scan mode is LR1110_WIFI_SCAN_MODE_FULL_BEACON).
The current documentation gives very exact looking formulas, so people expect them to be correct.

mverdy, 2020-11-18:
Thanks for the feedback @jamesl-dm, it is really appreciated!

If you spot any other suspect documentation issue don't hesitate to open a ticket.

sohaibqamar, 2022-01-03:
Hi @jamesl-dm 
Hope you are doing well.
Can you post a snippet of code for wifi scanning?
I am trying wifi passive scanning but have not been able to get any results so far. I have an open issue at the below link.
[https://github.com/Lora-net/lr1110_driver/issues/22](url)
Regards
Sohaib