Lora-net/lr1110_driver #8: Error from server: "Ignore invalid GNSS message payload"

gianni2996, 2020-12-17:
Hi, 
I am currently working on the GNSS part of LR1110.
 I succed in launching an autonomous scan and retreive from server the results with the position of the device. 
Now I am trying to launch an assisted scan: the scan is succesful, I can read the number of detected satellites and their ID, but when I send the json object to the server, in response I receive always this error:
'Ignore invalid GNSS message payload: 0e1b00b947134218a0382a2b476c46bb7a678c00'
 (this is for only 2 satellites detected, but the same is when i am outdoor and there are 10/11 satellites detected.) 
Reading the results of the assisted scan, when it detects some satellites the nav message is strange, it does not begin with 0101... (like with the autonomous), whereas when there are no satellites, as I expected it is 0x0 0x7.
The configuration of the system is the same that I set for the autonomous scan, what could be the problem?

mverdy, 2020-12-17:
Hi @gianni2996 ,

Indeed, the payload `0e1b00b947134218a0382a2b476c46bb7a678c00` does not look like a NAV message. The NAV message read from the LR1110 must start with `0101` (or `00` in case of failure like `0007` as you indicated when no satellite is detected). And when sending to the server, the first `01` is to be removed, so that the NAV message sent to the server must start with `01`.

Could you double check that the buffer you are using to store the NAV message from LR1110 is not corrupted on the host micro controller?
Could you also provide the version of the LR1110 you are using?
How are you transferring the NAV message from the device to the server?

gianni2996, 2020-12-17:
I solved the problem. The issue was that after reading the results size, before reading the results, I was reading the number of detected satellites, so probably the results were corrupted in some way. Now, reading the results immediatly after returned their size it works correctly. Thanks for the attention.

mverdy, 2020-12-17:
Thanks, I'll close this ticket as it is solved. Don't hesitate to open another one if needed.

closed 2020-12-17 by mverdy