Lora-net/lr1110_driver #14: Cannot obtain LR1110 firmware version

kvkc97, 2021-08-02:
I tried to get the firmware version of LR1110 using the command lr1110_bootloader_get_version() from lr1110_bootloader.c file.
As the response, I got 0 value in the firmware buffers. The value of stat register buffer[0] is coming correctly, so communication is ok. Can you help me solve this? The output for the command as displayed in COM port is attached.
[image]

matthieuantoine, 2021-08-02:
Hello,

I see that you already open a similar issue [there ](https://github.com/Lora-net/lr1110_modem_driver/issues/7)- is it a duplicate?

Would it be possible to get more information about your implementation? Can you also share some traces of signals between the LR1110 and the host MCU?

Thanks.

kvkc97, 2021-08-02:
Yes it is the same issue.

I have interfaced LR1110 IC with STM32WB55RGV6 controller. I have used STM32CubeIDE for programming. I have tested other commands using the interface with SPI and they have been successful (I have received CMD_OK response for them and have checked status register values). Only this command is failing.

matthieuantoine, 2021-08-02:
> I have tested other commands using the interface with SPI and they have been successful (I have received CMD_OK response for them and have checked status register values). Only this command is failing

Did you try another read command (like `lr1110_system_read_uid` ) to check if your implementation of the related HAL function is correct?

kvkc97, 2021-08-04:
The issue is solved. There was a connection error for busy pin due to which command was failed.
The current firmware version in LR1110 is 3.3 and the latest version available is 3.6. Can I know what are the differences between the two? Also is there any bug in version 3.3?

Currently I am not using TCXO in the circuit and it was mentioned in the datasheet that it id mandatory for GNSS scanning. Will the GNSS scan command fail in the absence of TCXO or just accuracy will be reduced?

I have implemented WiFi scanning without setting the RF frequency of sub-GHz radio. However i got Nbresults and MAC address. Is it not compulsory to set the frequency for WiFi scanning?

matthieuantoine, 2021-08-04:
> The issue is solved. There was a connection error for busy pin due to which command was failed.

That's great!

> The current firmware version in LR1110 is 3.3 and the latest version available is 3.6. Can I know what are the differences between the two? Also is there any bug in version 3.3?

Here are the places where you can find the requested information:
* https://github.com/Lora-net/radio_firmware_images/blob/master/lr1110/transceiver/README.md -> known limitations
* https://github.com/Lora-net/radio_firmware_images/blob/master/lr1110/transceiver/CHANGELOG.md -> changes between versions

> Currently I am not using TCXO in the circuit and it was mentioned in the datasheet that it id mandatory for GNSS scanning. Will the GNSS scan command fail in the absence of TCXO or just accuracy will be reduced?

The command will not fail but the results won't be as good as the one you can get with a TCXO.

> I have implemented WiFi scanning without setting the RF frequency of sub-GHz radio. However i got Nbresults and MAC address. Is it not compulsory to set the frequency for WiFi scanning?

No, it is not needed. A Wi-Fi scan will perform needed internal RF configurations.

kvkc97, 2021-08-04:
Thanks for the quick response. I an trying to implement GNSS Scanning currently and having some issues.
I have used the following commands for the GNSS scan.
1. GNSSSetConstellationToUse() to set GPS constellation. This command was successfully executed in the IC.
2. GnssSetMode() to set single scanning option. The value of Stat 1 and Stat2 registers after this command were 6 and 0 respectively. Is it a valid response? (Stat 2 register was never 0 for any other command)
3. GnssAutonmous() with the following arguments. The value of Stat1 register was 3. I crosschecked the implemented but couldnot find an error in it. The GNSS Done interrupt flag was set after this command but the size of result was obtained as 0.
[image]

I have attached the implemented code for 2. and 3. below. Could you please suggest what went wrong.
[image]

matthieuantoine, 2021-08-04:
Can you please close this ticket and open a new one with the question above so we keep things consistent and easy to find? Thanks!

kvkc97, 2021-08-04:
Sure

closed 2021-08-04 by kvkc97