Lora-net/lr1110_driver #15: Error in GNSS scanning

kvkc97, 2021-08-04:
I am trying to implement GNSS Scanning currently and having some issues.
I have used the following commands for the GNSS scan.

GNSSSetConstellationToUse() to set GPS constellation. This command was successfully executed in the IC.
GnssSetMode() to set single scanning option. The value of Stat 1 and Stat2 registers after this command were 6 and 0 respectively. Is it a valid response? (Stat 2 register was never 0 for any other command)
GnssAutonmous() with the following arguments. The value of Stat1 register was 3. I crosschecked the implemented but couldnot find an error in it. The GNSS Done interrupt flag was set after this command but the size of result was obtained as 0.
image
I have attached the implemented code for 2. and 3. below. Could you please suggest what went wrong.
image

kvkc97, 2021-08-22:
Can I get a solution for this issue.

gianni2996, 2021-08-22:
Hi @kvkc97, can I see also the code where you return the results of the scan? Thanks

kvkc97, 2021-08-22:
To check the results I have two functions, GnssGetNbSvDetected() and GnssGetResultSize(). The result size is returning value 2 while number of satellites is 0. I have attached the  implementing codes below.
[image]
GnssGetNbSvDetected() function
[image]
GnssGetResultSize() function
[image]

matthieuantoine, 2021-08-22:
Several questions:

- What is the version of the firmware you are using?
- What is the version of the driver you are using?
- Do you check the status returned by the functions before processing the out parameters?

kvkc97, 2021-08-23:
HW Version of LR1110 = 34
FW Version of LR1110 = 03.03

I checked the value of status register after each command once more now. After set_scan command, the status register values are 6 and  respectively. After scan command, the GNSS_DONE interrupt is  generated and stat1 value shows the last command was succesfully executed. Stat2 value is 19. The Stat1 and Stat2 values are 4 and 19 for all other commands. However, no results is obtained in the number of satellites detected.

gianni2996, 2021-08-23:
Supposing that you are using the modem version driver, you have to use the get_event_size() and get_event() functions on the interrupt firing, in order to retrieve the size and the results, for me it works.  Notice that if you call the get_event(), the function get_event_size() is already called inside the get_event(), so you don't have to call it twice or maybe you can have some problems as I experienced.

matthieuantoine, 2021-08-23:
@gianni2996 - according to the FW version shared by @kvkc97, he is running a transceiver.

Let me try to reproduce the exact same sequence with the same firmware version.

closed 2022-09-30 by smtc-bot