Lora-net/lr1110_driver #16: LR1110 LoRa transmission

kvkc97, 2021-08-05:
I have been to communicate LR1110 IC with a gateway with LoRaWAN protocol. After setting the parameters as per LR1110 User Manual, i used set_tx function to transmit. The packet is received at the gateway however there is an error occuring that PLL did not lock. I tried changing the frequency and used calibrate() with calibparam value 0x24 but error still exists. How to solve it?

Also could you please suggest a LoRaWAN Stack to be used for further development.

matthieuantoine, 2021-08-22:
Can you please share the configuration of the board you are using? Do you have a TCXO?

Concerning the LoRaWAN stack, you can have a look at https://github.com/Lora-net/LoRaMac-node

kvkc97, 2021-08-22:
[image]
[image]
[image]
[image]
I have attached the code snippet of LR1110 configuration. TCXO is not used in this case although provision is provided in the schematic. LR1110 interface in schematic is also attached.
[image]

matthieuantoine, 2021-08-22:
Where did you spot the error stating that the PLL did not lock?

Some comments:

- it is recommended to set the packet type before you configure the RF frequency
- the value assigned to `rbu` is incompatible with its type

kvkc97, 2021-08-23:
PLL lock error is obtained after set_tx function. I have checked for errors after each functions and no error was obtained. 

I changed the flow to call set packet type before configuring the RF frequency, but same error is obtained.

 I have also inserted MAC layer code from https://github.com/Lora-net/LoRaMac-node and send Join Request function also. The Request is correctly detected at the server and Accept is also generated however PLL error is still there.

kvkc97, 2021-09-03:
Hi,

Any update on this issue. Also, is there a need to switch to FS as intermediate mode while switch from TX to RX and vice versa while using auto tx rx mode.

kvkc97, 2021-09-07:
Can I have an update on this

matthieuantoine, 2021-09-07:
> Also, is there a need to switch to FS as intermediate mode while switch from TX to RX and vice versa while using auto tx rx mode.

No, there is not. The chip takes care of the state transitions.

matthieuantoine, 2021-09-07:
It seems that you introduced another topic with your last message - related to LoRaMAC node.

Can you please open an issue in the dedicated repository?

closed 2021-09-07 by kvkc97

MagnusErler, 2024-04-04:
@kvkc97 Can you share your solution for joining a Lora network. What I'm seeing from your code, is that you are setting the parameters to Lora and then just sending a command. When are you joining the network?