Lora-net/lr1110_driver #17: Functionality of random function

Pirakutu, 2021-08-05:
Hi guys,

I am using the LR1110 in a solution of ours and need the random command lr1110_system_get_random_number from lr1110_system.

Is this number supposed to be really random? At the moment, I only get 4 different values out of the function, which makes me think this is not a real source of randomness?

Thanks for your support.

Cheers

matthieuantoine, 2021-08-22:
Hi @Pirakutu,

We do not expect this number to be super random - but at least more random than just 4 different values. Let me check on my side.

Thanks.

matthieuantoine, 2021-08-29:
Hi @Pirakutu,

Can you please share the firmware version you are using?

Thanks.

Pirakutu, 2021-08-31:
The FW version of the LR1110? I will check today and pass you that info. So a bit more to the behavior, the first number generated after power up (i turn the chip on/off on runtime) is pretty random, after that, 4 different numbers show up. From these numbers 2 have the same 2 LSBs, which made it even more limited in usage, as the random number is used as DevNonce on the join message. For now i built a workaround using another source of randomness so I have the system running.

matthieuantoine, 2021-08-31:
FYI, I just did a simple test with LR1110 firmware 0x0306 (the latest one to date): request 255 random numbers in a row. I am not able to reproduce the issue you are seeing.

Pirakutu, 2021-09-01:
FW version is 0x0303. 
In LoRaMAC init is used as follows:

    Radio.Init( &MacCtx.RadioEvents );

    // Initialize the Secure Element driver
    if( SecureElementInit( EventSecureElementNvmCtxChanged ) != SECURE_ELEMENT_SUCCESS )
    {
        return LORAMAC_STATUS_CRYPTO_ERROR;
    }

    // Initialize Crypto module
    if( LoRaMacCryptoInit( EventCryptoNvmCtxChanged ) != LORAMAC_CRYPTO_SUCCESS )
    {
        return LORAMAC_STATUS_CRYPTO_ERROR;
    }

    // Initialize MAC commands module
    if( LoRaMacCommandsInit( EventCommandsNvmCtxChanged ) != LORAMAC_COMMANDS_SUCCESS )
    {
        return LORAMAC_STATUS_MAC_COMMAD_ERROR;
    }

    // Set multicast downlink counter reference
    if( LoRaMacCryptoSetMulticastReference( MacCtx.NvmCtx->MulticastChannelList ) != LORAMAC_CRYPTO_SUCCESS )
    {
        return LORAMAC_STATUS_CRYPTO_ERROR;
    }

    // Random seed initialization
    srand1( Radio.Random( ) );

    Radio.SetPublicNetwork( MacCtx.NvmCtx->PublicNetwork );
    Radio.Sleep( );

Then when a random number is needed inside LoRaMAC (for example when joining) in the function LoRaMacCryptoPrepareJoinRequest:

      SecureElementRandomNumber( &devNonce );

Which calls:

      Radio.Random( );

Radio.Random does the following:

    uint32_t RadioRandom( void )
    {
        uint32_t rnd  = 0;
        RadioStandby( );
        lr1110_system_get_random_number( &LR1110, &rnd );
        return rnd;
    }

Pirakutu, 2021-10-01:
Do you have any comment on this? Is this fixed on the newer version of the driver?

closed 2022-09-30 by smtc-bot