Lora-net/lr1110_driver #19: Error in integrating LoRaMAC node and LR1110 Driver

kvkc97, 2021-09-07:
I was testing for LoRa transmission with LR1110 IC in OTA mode using LoRaMAC code. The Join request is successfully sent to the server and Join accept is generated from the server and is passed to the gateway. The gateway also forwards the packet. However the packet is not received at the end device (custom board developed using STM32 controller). I have attached the Join request and join accept messages.
The IC is going to rx mode after transmitting but packet is not received. At time the IC goes to busy after TX mode and gets stuck. At times it goes to RX mode. A few times when in RX mode, interrupt for preamble detection is generated but no further packet is received and most of the times preamble detection also is not done. What would be the issue? Is it due to the PLL error generated.
Join request at server
[image]
Join accept at server
[image]
Join request and Join Request at gateway
[image]

matthieuantoine, 2021-09-07:
Can you please check that Rx windows are timely open? If you miss the beginning of the packet, the reception cannot be performed.

kvkc97, 2021-09-07:
Yes I checked that. The RX windows are opened correctly.

matthieuantoine, 2021-09-07:
> At time the IC goes to busy after TX mode and gets stuck

Can you please define `stuck`?

I think that a logical trace would help to catch issue.

kvkc97, 2021-09-07:
Before sending a command to the IC the busy pin values is checked to see if the IC is busy or not. After TX mode sometimes the busy pin value of the IC remains high. So IC is always busy and the program is stuck

kvkc97, 2021-09-10:
The firmware version in LR1110 is 0303. Is this version compatible with LR1110 Version1.0.0. If not, how can I change the firmware version of LR1110? Is it possible using SPI?

Pirakutu, 2021-09-10:
I have it running on version 0303 with the driver version 1.0.0, but i also had this busy pin problem. 

I rewrote the function to include a timeout so it doesnt get stuck for ever. However, i only had this happened to me when i turned the LR1110 on and off (took power away) between messages and only at certain times (not really reproducible). The cases where i left the the device powered from the beginning, were never a problem.

kvkc97, 2021-09-23:
I added the timeout and code does get stuck there. However, I still cannot receive the packet and PLL error is still there.

closed 2022-09-30 by smtc-bot