Lora-net/lr1110_driver #20: Lora transmission happens only in SF7 and SF8

kvkc97, 2021-09-17:
I am working on LR1110 in IN865 in ABP mode. Data transmission is successful in SF7 and SF8 but not in other spreading factors. The size of the packet is 17 bytes and adr is turned off. Frequency of transmission is 865062500. What can be issue for other spreading factors?

kvkc97, 2021-09-23:
Can anyone update this issue

matthieuantoine, 2022-01-10:
> I am working on LR1110 in IN865 in ABP mode

Are you referring to the [LoRaMAC node](https://github.com/Lora-net/LoRaMac-node) project?

kvkc97, 2022-01-12:
> > Are you referring to the [LoRaMAC node](https://github.com/Lora-net/LoRaMac-node) project?

Yes

matthieuantoine, 2022-01-12:
In this case, can you please open an issue in the LoRaMAC node project?

mluis1, 2022-01-12:
Please do not create an issue. Do create a new discussion topic under https://github.com/Lora-net/LoRaMac-node/discussions

closed 2022-02-23 by matthieuantoine