Lora-net/lr1110_driver #21: Usage example with LoRaMac stack?

Sensative-Lars-Mats, 2021-11-10:
Is there any example on how to use the LR1110 modem with the semtech LoRaMac stack?
Also using the GNSS functionality with the same?

matthieuantoine, 2021-11-10:
Hi @Sensative-Lars-Mats,

If you are referring to LoRaMAC-node, the LR1110 is supported from [v4.4.4](https://github.com/Lora-net/LoRaMac-node/blob/master/CHANGELOG.md#444---2020-05-26).

> Also using the GNSS functionality with the same?

Can you please elaborate?

Sensative-Lars-Mats, 2021-11-11:
It seems the examples on github only use the Basics Modem stack but we invested a lot on the LoRaMax stack which also supports this stack from 4.4 approximately. So we need some info on how to use the GNSS part of the driver.

Sensative-Lars-Mats, 2021-11-11:
Note: we expect to use the https://www.loracloud.com/documentation/geolocation?url=gettingstarted.html#gnss-based-location-estimates-for-lr1110-transceivers on the other end...

closed 2021-11-22 by Sensative-Lars-Mats