Lora-net/lr1110_driver #22: Passive Wi-Fi  Scanning

sohaibqamar, 2022-01-03:
Hi,
I am using the latest drivers of LR1110 with Zephyr for integration with NRF9160.
I'm attaching my lr1110_hal.c file. [lr1110_hal.pdf](https://github.com/Lora-net/lr1110_driver/files/7802121/lr1110_hal.pdf)
My NRF9160 is successfully communicating with LR1110 on SPI. 
I am getting some basic commands response like GetStatus, etc. But I am unable to get a correct response for Wifi scanning. 
Below is the code for scanning Wifi.
`
// Main File
void main(void)
{

    printk("LR1110 Wi-Fi Example\n");
	const uint8_t max_results = 10;
	const uint8_t nb_scan_per_channel = 10;
	const uint16_t timeout_in_ms = 70;
	const bool abort_on_timeout = false;
	lr1110_wifi_version_t wifi_ver;
	uint8_t nb_results = 0;
	setup_spim();
	test_SPIM();
	while (1) {
		printk("Scanning Wi-Fi...\n");
		lr_status = lr1110_hal_reset(dev);
		//channels      = 0x3FFF;  --> by default enable all channels		
		lr_status = lr1110_wifi_scan(dev, LR1110_WIFI_TYPE_SCAN_B_G_N,
									0x3FFF, LR1110_WIFI_SCAN_MODE_BEACON,
									max_results, nb_scan_per_channel,
									timeout_in_ms, abort_on_timeout);
		if(lr_status == LR1110_STATUS_OK) {
			printk("lr1110_wifi_scan OK \n");
		}
		else {
			printk("lr1110_wifi_scan ERROR!\n");
		}
		
		lr_status = lr1110_wifi_get_nb_results(dev, &nb_results);
		if(lr_status == LR1110_STATUS_OK) {
			printk("Num of Results: %d\n\n", (int)nb_results);
		}
		else {
			printk("NB Results ERROR!\n");
		}
		lr1110_wifi_basic_complete_result_t all_results[LR1110_WIFI_MAX_RESULTS] = {0};
		printk("All results created.\n");
		lr1110_wifi_read_basic_complete_results(dev, 0, 4, all_results);
		printk("After complete_results call.\n");
		for (int i = 0; i < nb_results; i++) {
			printk("%d :", i);
			//printk("\t%s :", all_results[i].mac_address);
			printk("\tRSSI : 0x%02X ", all_results[i].rssi);
			printk("\n");
		}
		printk("\n\n");
		k_sleep(K_MSEC(10000));
	}
}`

I am getting  _Num of Results: 0_ every time.
Please let me know if I'm making any mistakes in doing wifi scanning.
Do I need to add any additional initialization before calling lr1110_wifi_scan?
Waiting for the feedback.
Regards
Sohaib

mverdy, 2022-01-03:
Hi @sohaibqamar, it looks that the code is not waiting for the LR1110 to terminate the Wi-Fi scan before reading the results.

The reading of the results (with `lr1110_wifi_get_nb_results` and `lr1110_wifi_read_basic_complete_results`) must occur **after** the interruption `LR1110_SYSTEM_IRQ_WIFI_SCAN_DONE` has been raised by the LR1110. So the code must waits for this interruption to occur before fetching the results.

sohaibqamar, 2022-01-03:
> 

Hi @mverdy, thanks for the response.
How can I check  `LR1110_SYSTEM_IRQ_WIFI_SCAN_DONE` interrupt has occurred?
I am already checking the BUSY signal in the lr11110_hal.c read & write functions. Do I need to check this interrupt signal as well before calling  `lr1110_wifi_get_nb_results` ?

Regards
Sohaib

mverdy, 2022-01-03:
Well you are right, waiting for the busy pin should avoid to read before the scan is terminated. Could you check that reading the version information works correctly (with `lr1110_system_get_version`)?
Also, are you certain that there are Wi-Fi networks nearby? If there are few Wi-Fi signals, you may also consider increasing the value of `timeout_in_ms ` to something like 110 ms.

Concerning the wait for `LR1110_SYSTEM_IRQ_WIFI_SCAN_DONE`, you can poll the IRQ status with command `lr1110_system_get_irq_status`. When the scan is terminated, the device should return an IRQ mask with `LR1110_SYSTEM_IRQ_WIFI_SCAN_DONE` set.

sohaibqamar, 2022-01-03:
Hi @mverdy 
> Could you check that reading the version information works correctly (with `lr1110_system_get_version`)?

Yes, I have checked that the output is as in the below picture, I am doing this as below
`
// Code in main

        lr1110_system_version_t lrVer;
	lr_status = lr1110_system_get_version(dev, &lrVer);
	if(lr_status == LR1110_STATUS_OK) {
		printk("HW: %u, Type: %u, FW: %u\n\n", lrVer.hw, lrVer.type, (unsigned int)lrVer.fw);
	}
	else {
		printk("Error\n");
	}`

[image]

I am not sure whether I am getting the correct System Version or not, there is no working example of LR1110 available so I can compare my output with that. Also, there is no working example of Wi-Fi scanning available, even there is no proper workflow found in the documentation, I would really appreciate it if you can help me with this.

> Also, are you certain that there are Wi-Fi networks nearby?

Yes, there are almost 5-6 Wi-Fi networks available, I have even tried after turning ON my Phone's Hotspot.

mverdy, 2022-01-04:
Thanks for the feedback. It looks that the initialization steps of the LR1110 are missing. Look at [these lines](https://github.com/Lora-net/lr1110_evk_demo_app/blob/f46c2234c53f290a5e9fcedc41d177159d04eb68/embedded/device/src/device_transceiver.cpp#L52-L72) for the init steps.

sohaibqamar, 2022-01-05:
Hi, @mverdy 
Thanks a bundle, I was able to get Wi-Fi MAC Addresses after adding the below line in setup for LR1110.

`lr_status = lr1110_system_set_tcxo_mode(dev, LR1110_SYSTEM_TCXO_CTRL_3_3V, 500);`

closed 2022-01-05 by sohaibqamar